Z-Bolt/OctoScreen #324: dpkg-deb (subprocess): decompressing archive member: lzma error: compressed data is corrupt

stonemandave, 2021-11-07:
Selecting previously unselected package octoscreen.
(Reading database ... 92001 files and directories currently installed.)
Preparing to unpack octoscreen_2.7.4_armhf.deb ...
Unpacking octoscreen (2.7.4-1) ...
dpkg-deb (subprocess): decompressing archive member: lzma error: compressed data is corrupt
dpkg-deb: error: <decompress> subprocess returned error exit status 2
dpkg: error processing archive octoscreen_2.7.4_armhf.deb (--install):
 cannot copy extracted data for './usr/bin/OctoScreen' to '/usr/bin/OctoScreen.dpkg-new': unexpected end of file or stream
Errors were encountered while processing:
 octoscreen_2.7.4_armhf.deb

I can't install octoscreen. this is the error i get returned.

JeffB42, 2022-02-06:
I'm not able to reproduce this.  Does this still happen for you?

stonemandave, 2022-02-06:
I eventually wiped everything and uploaded a backup and the problem still persisted. I eventually found out it was a plugging causing the problem. Forgot which one. Deleted and problem was solved.

closed 2022-02-14 by JeffB42