Z-Bolt/OctoScreen #333: Screen update between CR-10S Pro and Octoscreen

PaulT133, 2022-01-27:
It appears that there is a 10+ second delay between Octoscreen and the screen on the printer. The delay hovers around 10-15/20 seconds. Can this be changed back to more of a real time update?

Thanks,
Paul

JeffB42, 2022-02-06:
Compare what OctoScreen displays with what OtcoPrint displays.  OctoScreen gets the printer's info via OctoPrint, and there usually is a slight delay.  Now, if OctoPrint matches your printer, and OctoScreen doesn't match OctoPrint, then there might be an issue.

PaulT133, 2022-02-07:
Hi Jeff,

Thank you for your reply. So Octoprint does match the printer display and it appears that Octoscreen is lagging to match Octoprint.It appears to be about 15-20 seconds behind. Previous versions lagged a second or two but this one seems very evident. Please let me know what else I can do to help.

JeffB42, 2022-02-16:
I'm in the middle of working on 2.8.  (and sorry, but no release estimate just now) and will look into this before I release it.