Z-Bolt/OctoScreen #334: OctoScreen Segmentation Faults at startup

D-e-n-t, 2022-02-07:
Trying to run OctoScreen with an Elegoo 3.5" LCD.
octoscreen service does not start.  X starts and stops, but OctoScreen does not get started.
Set logs per instructions from #199, but OctoScreen doesn't get to the point of opening the log file before it Segfaults.

strace -f shows:
gettimeofday({tv_sec=1644220433, tv_usec=729507}, NULL) = 0
futex(0xb6be0980, FUTEX_WAKE_PRIVATE, 2147483647) = 0
futex(0xb6be0980, FUTEX_WAKE_PRIVATE, 2147483647) = 0
statfs("/sys/fs/selinux", 0xbe8856e4)   = -1 ENOENT (No such file or directory)
statfs("/selinux", 0xbe8856e4)          = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/proc/filesystems", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0444, st_size=0, ...}) = 0
read(3, "nodev\tsysfs\nnodev\ttmpfs\nnodev\tbd"..., 1024) = 310
read(3, "", 1024)                       = 0
close(3)                                = 0
access("/etc/selinux/config", F_OK)     = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/proc/self/auxv", O_RDONLY) = 3
read(3, "\20\0\0\0\326\201\0\0", 8)     = 8
read(3, "\6\0\0\0\0\20\0\0", 8)         = 8
read(3, "\21\0\0\0d\0\0\0", 8)          = 8
read(3, "\3\0\0\0004\0\1\0", 8)         = 8
read(3, "\4\0\0\0 \0\0\0", 8)           = 8
read(3, "\5\0\0\0\f\0\0\0", 8)          = 8
read(3, "\7\0\0\0\0\20\356\266", 8)     = 8
read(3, "\10\0\0\0\0\0\0\0", 8)         = 8
read(3, "\t\0\0\09\233\5\0", 8)         = 8
read(3, "\v\0\0\0\0\0\0\0", 8)          = 8
read(3, "\f\0\0\0\0\0\0\0", 8)          = 8
read(3, "\r\0\0\0\0\0\0\0", 8)          = 8
read(3, "\16\0\0\0\0\0\0\0", 8)         = 8
read(3, "\27\0\0\0\0\0\0\0", 8)         = 8
read(3, "\31\0\0\0\344X\210\276", 8)    = 8
read(3, "\32\0\0\0\0\0\0\0", 8)         = 8
read(3, "\37\0\0\0\350_\210\276", 8)    = 8
read(3, "\17\0\0\0\364X\210\276", 8)    = 8
read(3, "\0\0\0\0\0\0\0\0", 8)          = 8
read(3, "", 8)                          = 0
close(3)                                = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=NULL} ---
+++ killed by SIGSEGV +++
Segmentation fault

D-e-n-t, 2022-02-07:
Using 2.7.4.  Get same issue on 2.7.0, 2.7.1 and 2.7.3.  Illegal instruction on 2.7.2.

D-e-n-t, 2022-02-09:
Turns out I was installing on a Raspberry Pi 1 B+, instead of the Pi 3 B+ that I intended.  It works fine on the Pi3B+

D-e-n-t, 2022-02-09:
One thing that should be corrected in all the guides is to install lightdm before installing octoscreen.  This allows the display manager configuration to come up automatically and is much easier than the process in the guide.

JeffB42, 2022-02-14:
Glad it's working for you now.

closed 2022-02-14 by JeffB42