Z-Bolt/OctoScreen #336: Minimum requirements Raspberry Pi

Juicer2002, 2022-02-15:
Sorry for asking i couldn't find it in FAQ. What is the minimum required RPB Pi ? Can the Raspberry Pi 2 still work for this job or is it to outdated? And can the Raspberry Pi 3B + accomplish the job?

Greets Juicer

Kyrunix, 2022-02-15:
Q: What CPUs does OctoScreen run on?

A: At this time, only the Raspberry Pi 4 and Raspberry Pi 3B are supported. Other CPUs like the Raspberry Pi Zero and BeagleBone are not supported. They might work, but if you encounter problems, it's up to up you to troubleshoot and work through issues you encounter.

Pi 2 is not supportet, Pi 3B+ should works. (I'm not sure but i think i made my tests of this project on a 3B+)

JeffB42, 2022-02-16:
@Juicer2002 it's not in the FAQ, but maybe I should add it.  The minimum requirements are mentioned in this setup guide:  https://github.com/Z-Bolt/OctoScreen/wiki/Setting-Up-OctoScreen-and-Your-Display#supported-hardware.  You can use a RPi 3B+ or a 4, but I would highly recommend using a RPi 4 - the performance difference is night & day.  Remember, you're running an instance of OctoPrint and OctoScreen on the same CPU.

closed 2022-02-16 by JeffB42