Z-Bolt/OctoScreen #338: Octoscreen on an Adafruit 3.5" TFT

BoomerT6, 2022-03-07:
Has anybody, ever, been able to install Octoscreen on an Adafruit 3.5" TFT? 
I have followed all the guides as faithfully as I could, but despite many attempts, I simply cannot get Octoscreen to display at all. 
I don't know if it is the Adafruit 3.5" TFT+Touchscreen (Product ID: 2441), or Octoscreen that is the problem. Or are they simply inconpatible? It is very possible that I am the problem, as I am not at home in the Linux enviroment.
I can follow step by step instructions, but I can't problem solve if the installation does not run smoothly.

baaygun, 2022-05-09:
I have used this to certain successs. will not work with Bullseye, though as Octoscreen does not support arm64 out of the box yet.

https://github.com/Rex516/Adafruit-3.5-Octoscreen-Setup/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen-from-Adafruit-%2AMay-6th-2021%2A

loykastj, 2022-07-18:
Follow this guide exactly and it will work: 
https://github-wiki-see.page/m/Rex516/Adafruit-3.5-Octoscreen-Setup/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen-from-Adafruit-%2AMay-6th-2021%2A

I just did it today.

BoomerT6, 2023-01-23:
Thanks for your help. I'm now up and running.

closed 2023-01-23 by BoomerT6