Z-Bolt/OctoScreen #339: Touchscreen issue

DannyGry, 2022-03-15:
Hi everyone, Im trying to download octoscreen to my raspbrry pi 3b+ and i have a usual 3.5'' lcd display, I do not now why, but i can not calibrate my touchscreen and it even does not work. In addition, it shows that my driver is ADS7846, but from documentation and back side of my display, I have a XPT2046 driver and I do not know how to change it, help me please ))
Here is some photoes and video
[image]

https://user-images.githubusercontent.com/86677716/158455041-c4955dba-9735-448e-8c1a-8add9c229bbe.MOV

felipebraz, 2022-05-20:
In my case (RPi 3B + 3.5" LCD) the touch never works... Any tips?

JeffB42, 2023-06-05:
Just what's in the Wiki...