Z-Bolt/OctoScreen #343: Initialization failed

bakaneko718, 2022-05-20:
I did a clean install a few times and the touch screen says: 

`initialization failed:
readConfig() failed, see log for errors`

I have a raspberry pi 4 and a hyper pixel 4 that I am trying to set up. 

' sudo dpkg -i octoscreen_2.7.4_armhf.deb
(Reading database ... 56000 files and directories currently installed.)
Preparing to unpack octoscreen_2.7.4_armhf.deb ...
Unpacking octoscreen (2.7.4-1) over (2.7.4-1) ...
Setting up octoscreen (2.7.4-1) ...
Job for octoscreen.service failed because a timeout was exceeded.
See "systemctl status octoscreen.service" and "journalctl -xe" for details.'

This has been what I have been getting. in putty. After rebooting I get the same error as before

bakaneko718, 2022-05-22:
I have retried a clean install and followed Chris' basement video step by step. i attempted to do a docker install, but got nothing from that. I saw there was another issue posted and that pointed me to a 3.5 screen, but that didn't help either.

DiogoBasto13, 2022-05-22:
I have se same problem, been trying many stuff, not working... And getting this screen.
[image]

JeffB42, 2022-05-23:
Well, what does the log file report?

bakaneko718, 2022-05-23:
I can't find the file. I go to where it should be and it isn't there. Unless I'm looking in the wrong place.

DiogoBasto13, 2022-05-23:
> I can't find the file. I go to where it should be and it isn't there. Unless I'm looking in the wrong place.

Same thing with mine

bakaneko718, 2022-05-24:
When I look through the config I see there is a thing asking where I want the file to be, but changing that didn't work either.

DiogoBasto13, 2022-05-27:
I think it might be related with the new raspbian, even Touch_UI wont work either.

tjwitman, 2022-06-02:
I believe DiogoBasto is right!  I tried to upgrade one of my pi's that I use on my printers and used the new Raspi imager.  Got the Octoprint to work on my laptop but nothing I try can get Octoscreen to work.  I use the deb files on the Octoscreen page with and without the initial sequence above the deb file.  There is a /esc/octoscreen/config folder but nothing in it to configure the program to accept my Pi original 7" screen.  I just go back to the pi showing pi login:

Something is screwed up.

tjwitman, 2022-07-01:
> I had the same issue initialization failed: readConfig() failed... under OctoScreen 2.7.4.
> For me it worked to set the correct path of the OctoPrint config file in /esc/octoscreen/config
> OCTOPRINT_CONFIG_FILE=/home/USERNAME/.octoprint/config.yaml
> Make sure that USERNAME is the username Octoprint is running. There is no user pi on newer Raspberry Pi OS.
> Attached are two debug-level logs. log_failed.txt is without OCTOPRINT_CONFIG_FILE set and log_success.txt is with correct path to OctoPrint config file.
>
> log_failed.txt
> log_success.txt

Thanks a lot.  I'll give it a try!!!
Tom

JohnnyPrimus, 2022-09-29:
@Darthyson Thanks a ton, this worked for me.

peterrlc, 2023-01-17:
I had to make certain that these variables had values in them as follows

OCTOPRINT_CONFIG_FILE=/home/USERNAME/.octoprint/config.yaml
OCTOSCREEN_LOG_FILE_PATH=/home/USERNAME/logs/logfile.txt
OCTOSCREEN_LOG_LEVEL=Debug
OCTOSCREEN_RESOLUTION=800x480
DISPLAY_CURSOR=True

where USERNAME is the login name for your RPi

logfile.txt did not exist on my installation. Once I set that up it worked.

nhegde24, 2023-08-26:
> I had the same issue `initialization failed: readConfig() failed...` under OctoScreen 2.7.4. For me it worked to set the correct path of the OctoPrint config file in `/etc/octoscreen/config` `OCTOPRINT_CONFIG_FILE=/home/USERNAME/.octoprint/config.yaml` Make sure that `USERNAME` is the username Octoprint is running. There is no user pi on newer Raspberry Pi OS. Attached are two debug-level logs. `log_failed.txt` is without `OCTOPRINT_CONFIG_FILE` set and `log_success.txt` is with correct path to OctoPrint config file.
> 
> [log_failed.txt](https://github.com/Z-Bolt/OctoScreen/files/9021445/log_failed.txt) [log_success.txt](https://github.com/Z-Bolt/OctoScreen/files/9021450/log_success.txt)

Hi, I am trying to update the config file in the folder you mentioned however, when i open the config file in notepad and update the username and try to save it, it give me the following error - "Error #3(/etc/octoscreen/config): Permission denied " 

Appreciate if you could assist on this.

[image]