Z-Bolt/OctoScreen #360: no octoprint presets loaded by octoscreen

FGOD1983, 2023-03-03:
Somehow, since a few weeks I am unable to see the presets from Octoprint in Octoscreen. I can still heat up with the increase temperature, but hitting the bed/hotend or going to temperature and then presets, there is nothing happening or to be seen.

I also tried adding a new preset in Octoprint, but that too is not seen in Octoscreen

the-ress, 2023-06-11:
I had the same issue - it was caused by OctoPrint taking over 3 seconds to respond and OctoScreen gave up.
#347 fixes the issue (unless yours is caused by something else).

FGOD1983, 2023-06-12:
Oh wow, I'll try to update it today and see if that fixes it, thx!

FGOD1983, 2023-06-12:
Thx! That did fix it indeed!
[image]

JeffB42, 2023-06-20:
https://github.com/Z-Bolt/OctoScreen/pull/347 was merged in, so this should be fixed now.

closed 2023-06-20 by JeffB42