Z-Bolt/OctoScreen #361: start octoscreen as a dependency of octoprint.service

c0psrul3, 2023-03-29:
currently, octoscreen.service does not start on startup.
systemd unit for octoscreen should define octoprint.service in WantedBy list

Gilligan8, 2023-05-23:
The work around that I implemented was to run the service start command (service octoscreen start &) at the end of the rc.local script before it exits.

It's working so far until a proper fix is out.

c0psrul3, 2023-05-24:
I was resolving a bug, while reporting it.  my fix is in line with systemd
service dependency patterns.

Gilligan8, 2023-05-31:
Oooh, so I would need to recompile with the WantedBY option?

I completely read your post incorrectly as a "request" not a "fix".

c0psrul3, 2023-05-31:
yea, my PR fixes the service start dependency completely within systems.
super simple

Gilligan8, 2023-05-31:
I was just trying to implement this and forgive my ignorance but I'm not sure where to run this.

I pretty much followed this guide:
https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen

I tried going back into the xf86-video-fbturbo dir and using that make command but it says "no rule to make target 'octoscreen.service'.

Clearly I'm doing something wrong... but as I stated, I'm kind of ignorant here.

c0psrul3, 2023-05-31:
sry, can't help. I am using rpi 7" screen, which worked out of the box for
me.  only needed to install the package.
🫤

JeffB42, 2023-06-05:
The 2.8.0 release should address most stability issues.