Z-Bolt/OctoScreen #365: Octoscreen missing

MPFG1, 2023-05-11:
Hi guys first time poster at Github here.

Problem: I had installed OctoScreen successfully before on my Raspberry Pi 3B+ (and Raspberry display v1.1).  I recently had to wipe and reinstall everything, and now... issues.  

When I run: 

_'sudo dpkg -i octoscreen_2.7.4_armhf.deb'_

I get:

_dpkg: dependency problems prevent configuration of octoscreen:
 octoscreen depends on libgdk-pixbuf2.0-0 (>= 2.25.2); however:
  Package libgdk-pixbuf2.0-0 is not installed.

dpkg: error processing package octoscreen (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 octoscreen_

[Octoscreen issue.txt](https://github.com/Z-Bolt/OctoScreen/files/11447911/Octoscreen.issue.txt)

Full printout attached.  Please help me figure out what needs to be done, in simple non-coding terms :)

dan0412, 2023-05-29:
I am having this same very error (on RPi 3b + Pi Foundation Touchscreen).I had a working install for years until I had a need to reinstall yesterday. It would be helpful to understand how to fix this.

MPFG1, 2023-05-30:
I found a guy on Fiverr to help me solve this, and here’s what he did:

I downloaded the SD image from the official OctoPi website.
I used BalenaEtcher to flash the SD image to the SD card. (Raspberry Pi Imager sometimes gave me trouble.)
After flashing, I set the Wi-Fi settings using the "wpa_supplicant.conf" file and added the "ssh" file to enable SSH.
Once the Raspberry Pi booted up, I ran the usual commands "sudo apt update" and "sudo apt upgrade" to update and upgrade the system.
At this point, I set up my 3D printer in OctoPi because some recommend setting up the 3D printer before installing OctoScreen.
Then, I installed the necessary dependencies using the following command: "sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils".
After that, I downloaded and installed OctoScreen using the following commands:

wget https://github.com/Z-Bolt/OctoScreen/releases/download/v2.7.4/octoscreen_2.7.4_armhf.deb
sudo dpkg -i octoscreen_2.7.4_armhf.deb
However, I encountered the same error you mentioned, which stated that "octoscreen" depends on "libgdk-pixbuf2.0-0" (version 2.25.2 or higher), but it was not installed.
To fix this, I ran the command "sudo apt-get install --fix-broken" to resolve all the dependency problems.
Once the dependencies were fixed, I ran the command "sudo dpkg -i octoscreen_2.7.4_armhf.deb" again to install OctoScreen.

dan0412, 2023-05-30:
Thanks - I'll try this. I eventually installed OctoDash - much better UI, features, etc. in IMHO.

closed 2023-06-05 by JeffB42