Z-Bolt/OctoScreen #366: Help, there are no button icons. What's wrong?

Aleks-MO, 2023-05-19:
Help, there are no button icons. What's wrong? 
[image]
The question was asked here, there is no solution.((((   https://github.com/bigtreetech/CB1/issues/121

JeffB42, 2023-06-05:
I suggest turing on logging and see if there are any errors.  See the Wiki for info on logging.

Aleks-MO, 2023-06-05:
@JeffB42  
204" file=logger.go func=logger.Error line=164
time="2023-06-05T13:20:36Z" level=error msg="                                Client.handleResponse() - StatusCode is 204" file=logger.go func=logger.Error line=164
time="2023-06-05T13:20:56Z" level=error msg="                                Client.handleResponse() - StatusCode is 204" file=logger.go func=logger.Error line=164
time="2023-06-05T13:21:16Z" level=error msg="                                Client.handleResponse() - StatusCode is 204" file=logger.go func=logger.Error line=164
time="2023-06-05T13:21:36Z" level=error msg="                                Client.handleResponse() - StatusCode is 204" file=logger.go func=logger.Error line=164
time="2023-06-05T13:21:56Z" level=error msg="                                Client.handleResponse() - StatusCode is 204" file=logger.go func=logger.Error line=164
time="2023-06-05T13:22:16Z" level=error msg="                                Client.handleResponse() - StatusCode is 204" file=logger.go func=logger.Error line=164
time="2023-06-05T13:23:38Z" level=error msg="                Client.handleResponse() - StatusCode is 204" file=logger.go func=logger.Error line=164
time="2023-06-05T13:23:57Z" level=debug msg=- file=logger.go func=logger.Debug line=137
time="2023-06-05T13:23:57Z" level=debug msg=- file=logger.go func=logger.Debug line=137
time="2023-06-05T13:23:57Z" level=debug msg="entering OctoScreen - main.init()" file=logger.go func=logger.TraceEnter line=101
time="2023-06-05T13:23:57Z" level=debug msg="    entering main.readConfig()" file=logger.go func=logger.TraceEnter line=101
time="2023-06-05T13:23:57Z" level=info msg="        Path to OctoPrint's config file: \"/home/biqu/.octoprint/config.yaml\"" file=logger.go func=logger.Infof line=150
time="2023-06-05T13:23:57Z" level=info msg="        main.readConfig() - server host is: \"localhost\"" file=logger.go func=logger.Infof line=150
time="2023-06-05T13:23:57Z" level=info msg="        main.readConfig() - server port is: 5000" file=logger.go func=logger.Infof line=150
time="2023-06-05T13:23:57Z" level=debug msg="    leaving main.readConfig()" file=logger.go func=logger.TraceLeave line=115
time="2023-06-05T13:23:57Z" level=debug msg="    entering main.setApiKey()" file=logger.go func=logger.TraceEnter line=101
time="2023-06-05T13:23:57Z" level=debug msg="        main.setApiKey() - APIKey is \"7BB---061\"" file=logger.go func=logger.Debugf line=141
time="2023-06-05T13:23:57Z" level=debug msg="    leaving main.setApiKey()" file=logger.go func=logger.TraceLeave line=115
time="2023-06-05T13:23:59Z" level=error msg="                        Client.doRequest() - this.httpClient.Do() returned an error: \"Get http://localhost:5000/api/connection: dial tcp [::1]:5000: getsockopt: connection refused\"" file=logger.go func=logger.LogError line=121
time="2023-06-05T13:23:59Z" level=error msg="                    Client.doJsonRequest() - this.doRequest() returned an error: \"Get http://localhost:5000/api/connection: dial tcp [::1]:5000: getsockopt: connection refused\"" file=logger.go func=logger.LogError line=121
time="2023-06-05T13:23:59Z" level=error msg="                ConnectionRequest.Do() - client.doJsonRequest(GET) returned an error: \"Get http://localhost:5000/api/connection: dial tcp [::1]:5000: getsockopt: connection refused\"" file=logger.go func=logger.LogError line=121
time="2023-06-05T13:23:59Z" level=error msg="            ui.verifyConnection() - Broke into the else condition because ConnectionRequest.Do() returned an error returned an error: \"Get http://localhost:5000/api/connection: dial tcp [::1]:5000: getsockopt: connection refused\"" file=logger.go func=logger.LogError line=121
time="2023-06-05T13:24:21Z" level=error msg="                                Client.handleResponse() - StatusCode is 204" file=logger.go func=logger.Error line=164
time="2023-06-05T13:24:39Z" level=error msg="                                Client.handleResponse() - StatusCode is 204" file=logger.go func=logger.Error line=164

the-ress, 2023-06-11:
I had this issue and solved it by installing `librsvg2-common`

Aleks-MO, 2023-06-12:
@the-ress  Good afternoon,  **librsvg2-common** - installed , the solution to the problem is not possible in this.((( This problem has been going on for a long time... https://github.com/Z-Bolt/OctoScreen/issues/288

JeffB42, 2023-06-20:
@Aleks-MO did you follow the set up instructions, and run `sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils

The images in /images/logos are PNG files. The images in /images are SVGs. Try installing librsvg (https://gitlab.gnome.org/GNOME/librsvg) though, SVG image support should have been installed when you installed those other libraries.

Seems like this is a duplicate of https://github.com/Z-Bolt/OctoScreen/issues/288

Aleks-MO, 2023-06-20:
Yes, I installed everything.

I have it like this, for 32 and 64 bit...
librsvg2-2/oldstable,now 2.50.3+dfsg-1 arm64 [installed]
librsvg2-2/oldstable,now 2.50.3+dfsg-1 i386 [installed,automatic]
librsvg2-bin/oldstable,now 2.50.3+dfsg-1 arm64 [installed]
librsvg2-common/oldstable,now 2.50.3+dfsg-1 arm64 [installed]
librsvg2-common/oldstable,now 2.50.3+dfsg-1 i386 [installed]
librsvg2-dev/oldstable,now 2.50.3+dfsg-1 arm64 [installed]

It looks like this version of the problem is https://github.com/Z-Bolt/OctoScreen/issues/288
Does this mean that everything will not work correctly?

JeffB42, 2023-07-04:
_"Does this mean that everything will not work correctly?"_

If I recall correctly, you need to uninstall the 64-bit version of libsvg and then install the 32-bit version.  (instructions are in that link)

If that doesn't work, you will most likely need to start again, and use the 32-bit version of OctoPrint.

closed 2023-07-04 by JeffB42