Z-Bolt/OctoScreen #369: octoscreen wont istall with deb package or source install

Thommie2000, 2023-06-21:
the first time i tried installing it, it worked perfectly fine. but when i shut it down and booted it the next day again, it didnt load the ui. I installed the octopi software many times again, but now it wont even install, I now tried it with the source install but after i cloned the page it wont make a build it keeps saying: make: *** No rule to make target 'build'.  Stop.

Thommie2000, 2023-06-21:
and when i install it with package and do this command `sudo dpkg -i octoscreen_2.7.4_armhf.deb` i get this: Selecting previously unselected package octoscreen.
(Reading database ... 60432 files and directories currently installed.)
Preparing to unpack octoscreen_2.7.4_armhf.deb ...
Unpacking octoscreen (2.7.4-1) ...
dpkg: dependency problems prevent configuration of octoscreen:
 octoscreen depends on libgdk-pixbuf2.0-0 (>= 2.25.2); however:
  Package libgdk-pixbuf2.0-0 is not installed.

dpkg: error processing package octoscreen (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 octoscreen

the-ress, 2023-06-21:
You can either run `sudo apt-get -f install` after dpkg to install the dependencies, or just run `sudo apt install ./octoscreen_2.7.4_armhf.deb`.

https://askubuntu.com/questions/40011/how-to-let-dpkg-i-install-dependencies-for-me