Z-Bolt/OctoScreen #370: Octoscreen fails to start on fresh install of Octoprint

tabjohitol, 2023-07-14:
Hello,

I recently re-installed Octoprint on my  Raspberry Pi 4. Octoscreen and Octorprint  had been working just fine once everything was up, but I was having to hit enter on the console (attached a keyboard) upon power-on because I was getting a locked root account message that I couldn't easily fix. I figured a quick reinstall would be the cleanest fix, and followed the instructions from the [wiki ](https://community.octoprint.org/t/how-can-i-find-out-the-version-of-octoprint-or-octopi-i-am-running/204) that had previously worked flawlessly for me.

Everything works fine, I get no errors throughout the install process.  However upon installing Octoscreen 2.7.4, Octoscreen fails to start and I get the following message:
```
Jul 13 20:50:29 jeeg-octopi xinit[1104]: X.Org X Server 1.20.11
Jul 13 20:50:29 jeeg-octopi xinit[1104]: X Protocol Version 11, Revision 0
Jul 13 20:50:29 jeeg-octopi xinit[1104]: Build Operating System: linux Raspbian
Jul 13 20:50:29 jeeg-octopi xinit[1104]: Current Operating System: Linux jeeg-octopi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64
Jul 13 20:50:29 jeeg-octopi xinit[1104]: Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=533 bcm2708_fb.fbswap=1 smsc95xx.macaddr=E4:5F:01:2F:7A:62 vc_mem.mem_base=0x3ec00000 vc>
Jul 13 20:50:29 jeeg-octopi xinit[1104]: Build Date: 30 March 2023  01:46:14PM
Jul 13 20:50:29 jeeg-octopi xinit[1104]: xorg-server 2:1.20.11-1+rpt3+deb11u6 (https://www.debian.org/support)
Jul 13 20:50:29 jeeg-octopi xinit[1104]: Current version of pixman: 0.40.0
Jul 13 20:50:29 jeeg-octopi xinit[1104]:         Before reporting problems, check http://wiki.x.org
Jul 13 20:50:29 jeeg-octopi xinit[1104]:         to make sure that you have the latest version.
Jul 13 20:50:29 jeeg-octopi xinit[1104]: Markers: (--) probed, (**) from config file, (==) default setting,
Jul 13 20:50:29 jeeg-octopi xinit[1104]:         (++) from command line, (!!) notice, (II) informational,
Jul 13 20:50:29 jeeg-octopi xinit[1104]:         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
Jul 13 20:50:29 jeeg-octopi xinit[1104]: (==) Log file: "/var/log/Xorg.0.log", Time: Thu Jul 13 20:50:29 2023
Jul 13 20:50:29 jeeg-octopi xinit[1104]: (==) Using config file: "/etc/X11/xorg.conf"
Jul 13 20:50:29 jeeg-octopi xinit[1104]: (==) Using config directory: "/etc/X11/xorg.conf.d"
Jul 13 20:50:29 jeeg-octopi xinit[1104]: (==) Using system config directory "/usr/share/X11/xorg.conf.d"
Jul 13 20:50:29 jeeg-octopi xinit[1105]: modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/6.1.21-v8+
Jul 13 20:50:30 jeeg-octopi xinit[1104]: /usr/lib/xorg/Xorg: symbol lookup error: /usr/lib/xorg/modules/drivers/fbturbo_drv.so: undefined symbol: shadowUpdatePackedWeak
```
I tried searching but failing to find a solution.  It looks like I am failing to load some necessary module? Can someone please please point me in the right direction or perhaps let me know what other diagnostics I can perform to figure out what the issue might be?

Appreciate any help!

tabjohitol, 2023-07-14:
Thanks, just tried your build and it looks like the same error. 

Would it be advisable to go back to Octopi 0.18 just to get the OctoScreen functionality working for now?

the-ress, 2023-07-14:
I tried to reproduce the issue, but I currently don’t have any RPi with me. (The closest I could do was an ARM64 Debian Bullseye VM in cloud, and it worked there.)

After googling the error message, I’m not entirely sure it’s the actual error. If you edit `/etc/octoscreen/config` and set `OCTOSCREEN_LOG_FILE_PATH` to something, does that file get created? If yes, does it contain anything interesting? (Also did you enter all the required settings in `/etc/octoscreen/config`?)

tabjohitol, 2023-07-14:
When I googled the errors, I saw that the g2d_23 error was probably benign, so I think that might be ok to ignore and it's the undefined symbol: shadowUpdatePackedWeak portion that's problematic?

I did try setting `OCTOSCREEN_LOG_FILE_PATH` to a value, but the file never gets created when attempting to start octoscreen.

This is what I have in `/etc/octoscreen/config`

```
 cat /etc/octoscreen/config  | grep -Ev "^#|^$"
OCTOPRINT_HOST=http://localhost:5000
OCTOPRINT_APIKEY=BD9B4xxxxxxxxxxxxxxxxxxxxxxxxxxxxx
OCTOSCREEN_STYLE_PATH=/opt/octoscreen/styles/z-bolt/
OCTOPRINT_CONFIG_FILE=/home/pi/.octoprint/config.yaml
OCTOSCREEN_LOG_FILE_PATH=/home/pi/logs/octoscreen.log
OCTOSCREEN_LOG_LEVEL=Error
OCTOSCREEN_RESOLUTION=800x533
```

the-ress, 2023-07-23:
I tested OctoScreen on the latest OctoPI yesterday and it worked fine. Then I noticed you fixed the [wiki](https://github.com/Z-Bolt/OctoScreen/wiki/Installing-OctoScreen-with-a-3.5%22-480x320-TFT-screen) link.

Now I think it's the LCD driver. The `waveshare/LCD-show` repo doesn't look well maintained and likely doesn't work with Bullseye. Looking through the issues there, I found someone recommending to use https://github.com/goodtft/LCD-show instead (https://github.com/waveshare/LCD-show/issues/56).

fargodwe, 2024-03-15:
The waveshare lcd-show I think is almost, if not completely, the same as that found under the goodtft github.  I used both with my generic tft and they both worked.  For the error message you showed, I found that I had to (1) be sure via raspi-config that the default (via system option) is the 2nd option - auto login to command prompt (2) use rc.local and add the systemctl entry to start the octoscreen service.  That's worked for me anyway.