Z-Bolt/OctoScreen #371: Issue with screen rotation

Simgy, 2023-08-21:
Hello,

I installed last version of Raspberry OS with an hyperpixel 4.0 touch.
I set the hyperpixel and touchscreen of my raspberry in landscape position with : 
- `dtparam=rotate=90,touchscreen-swapped-x-y,touchscreen-inverted-y`

but when I install octoscreen, the window of octoscreen is in portait mode.

I tried lot of solution with no result.
Do you have this type of problem? how did you solve it?
[image]

treakifo, 2023-09-19:
For what it's worth, I'm having the same problem, also using a Hyperpixel 4.0 touch and config.txt setting. The console text at boot appears in the correct orientation, but OctoScreen needs to be rotated 90 degrees. It was working on the same hardware from an earlier version of the OctoPi OS image, but I stupidly decided to rebuild with the current version.

fargodwe, 2024-03-15:
Did you try a 180 rotation?

treakifo, 2024-03-17:
I actually got it working! From Pimoroni's Hyperpixel driver page (https://github.com/pimoroni/hyperpixel4) the section entitled "Totally Manual Rotation", I was able to use the Rotation on the Fly instructions to rotate the screen the the correct value. The command stores the rotation you set and no additional manual config changes are needed.

I tried reproducing the steps on another Raspberry Pi 4 with another Hyperpixel, but I couldn't get it to work again. I finally just cloned the SD card I used to build the first Pi and that finally worked.

JeffB42, 2024-07-12:
Thanks @treakifo , I'm going to close this issue now.

closed 2024-07-12 by JeffB42