Z-Bolt/OctoScreen #372: Where are OctoScreen settings files located???

Renegade82, 2023-08-31:
Linux deploy + Linux Buster. 
OctoScreen: octoscreen_2.7.4_armhf. 
When configuration files are changed, nothing happens, Octoscreen does not react in any way. Even deleted them, the reaction is also zero. When run, it displays an error in the path to the theme display style. Changing /esc/octoscreen/config does nothing, absolutely. So where does it get the settings from!?
(The octoscreen_2.5-1_armhf version partially works, but there is no reaction to changes in the config file this way)

Renegade82, 2023-08-31:
``login as: pi
pi@172.20.10.5's password:
Linux localhost 3.4.0 #1 SMP PREEMPT Mon Jun 9 14:10:26 CST 2014 armv7l
Debian GNU/Linux 10 (buster) [running via Linux Deploy]
Last login: Wed Aug 30 23:45:17 2023 from 172.20.10.3
pi@localhost:~$ export DISPLAY=:0
pi@localhost:~$ OctoScreen
2023/08/31 10:13:58 logger.init() - logFilePath is was not defined.  Now using just the standard console output.
2023/08/31 10:13:58 logger.SetLogLevel() - the log level is now set to: debug
DEBU[0000] -                                             file=logger.go func=logger.Debug line=137
DEBU[0000] -                                             file=logger.go func=logger.Debug line=137
DEBU[0000] entering OctoScreen - main.init()             file=logger.go func=logger.TraceEnter line=101
DEBU[0000]     entering main.findConfigFile()            file=logger.go func=logger.TraceEnter line=101
DEBU[0000]         entering main.doFindConfigFile()      file=logger.go func=logger.TraceEnter line=101
ERRO[0000]             main.doFindConfigFile() - Stat() returned an error  file=logger.go func=logger.LogError line=123
DEBU[0000]         leaving main.doFindConfigFile(), returning path  file=logger.go func=logger.TraceLeave line=115
INFO[0000]         main.findConfigFile() - doFindConfigFile() found a file  file=logger.go func=logger.Info line=146
DEBU[0000]     leaving main.findConfigFile(), returning the file  file=logger.go func=logger.TraceLeave line=115
DEBU[0000]     entering main.readConfig()                file=logger.go func=logger.TraceEnter line=101
INFO[0000]         Path to OctoPrint's config file: "/home/pi/.octoprint/config.yaml"  file=logger.go func=logger.Infof line=150
INFO[0000]         main.readConfig() - server host is: "localhost"  file=logger.go func=logger.Infof line=150
INFO[0000]         main.readConfig() - server port is: 5000  file=logger.go func=logger.Infof line=150
DEBU[0000]     leaving main.readConfig()                 file=logger.go func=logger.TraceLeave line=115
DEBU[0000]     entering main.setApiKey()                 file=logger.go func=logger.TraceEnter line=101
DEBU[0000]         main.setApiKey() - APIKey is empty, now using cfg.API.Key  file=logger.go func=logger.Debug line=137
DEBU[0000]         main.setApiKey() - APIKey is "ED2---79D"  file=logger.go func=logger.Debugf line=141
DEBU[0000]     leaving main.setApiKey()                  file=logger.go func=logger.TraceLeave line=115
2023/08/31 10:13:58 logger.SetLogLevel() - the log level is now set to: error
2023/08/31 10:13:58 main.SetLogLevel() - logLevel is now set to: "error"`

JeffB42, 2023-09-01:
This information is in the home page: https://github.com/Z-Bolt/OctoScreen
Also see https://github.com/Z-Bolt/OctoScreen

closed 2023-09-01 by JeffB42

Renegade82, 2023-09-01:
What is not clear in my post? Default config not working! And do not send me to read something that has long been read and verified. In the trash this program.

FloodJ, 2023-09-02:
@Renegade82 - you asked a question - "where are the octoscreen files located" - and Jeff answered correctly. If you want the answer to another question then it would be best to ask it - politely.