Z-Bolt/OctoScreen #377: octoscreen seems to require a user account named pi, but that requirement is not documented (I beleive)

Dweinbach, 2024-04-08:
I was unable to get octoscreen to run on my fresh octopi install, no matter what I tried. followed all the directions, videos, etc. - it would install but the service would not be configured.

When i install pi images, i change the default username to dave from pi, and have not had a problem before, but it appears that octoscreen expects a user named pi to exist.

This should either be documented, an override provided, or both.

JeffB42, 2024-04-16:
Hmm... it shouldn't be dependent on being logged in with a "pi" account.  I searched the codebase - did you update `OCTOPRINT_CONFIG_FILE` in the config file (`debian/local/octoscreen/config`) to point to a path with `dave` in it?
(this is documented in `README.md`)

ArcAIN6, 2024-05-05:
octoscreen requires a login to run at boot. did you run raspi-config to set the boot up login?

[image]

[image]

[image]

As you can see, the auto-loging for me is not "pi" as i set a different name on that raspberry pi

ItsCharged, 2024-06-05:
> I was unable to get octoscreen to run on my fresh octopi install, no matter what I tried. followed all the directions, videos, etc. - it would install but the service would not be configured.
>
> When [...]

If you are talking about the install process and venv, usually it is a changing of the username there

JeffB42, 2024-07-12:
Looks like this isn't an issue anymore.  Thanks @ItsCharged .  I'm going to close this issue.

closed 2024-07-12 by JeffB42