Z-Bolt/OctoScreen #378: Octoscreen not loading on raspberry pi 5

ingo916, 2024-05-03:
I am running the official Raspberry Pi 7" touchscreen with a raspberry pi 4B works great!
I removed the pi 4 and installed a raspberry pi 5
Installed octoprint using the octoprint_deploy https://github.com/paukstelis/octoprint_deploy
octoprint boots up and I can access the web UI. 

installed required packages  using: sudo apt-get install libgtk-3-0 xserver-xorg xinit x11-xserver-utils
installed OctoScreen using both methods
wget https://github.com/Z-Bolt/OctoScreen/releases/download/v2.7.4/octoscreen_2.7.4_armhf.deb
sudo dpkg -i octoscreen_2.7.4_armhf.deb

and
wget https://github.com/Z-Bolt/OctoScreen/releases/download/v2.7.4/octoscreen_2.7.4_armhf.deb
sudo dpkg -r octoscreen
sudo dpkg -i octoscreen_2.7.4_armhf.deb
sudo reboot now

screen stays at the raspberrypi login: prompt

ingo916, 2024-05-03:
Running
Raspberry Pi OS Lite
Release date: March 15th 2024
System: 64-bit
Kernel version: 6.6
Debian version: 12 (bookworm)

ArcAIN6, 2024-05-05:
you're at login because you didn't run raspi-config to set it to login automatically

ingo916, 2024-05-05:
I did do that step

Ingo Azarvand
IT Manager
Direct 209-944-0836 x151
Cell 209-470-2565

4848 Frontier Way Ste 100 Stockton CA 95215
Mailing: PO Box 31330 Stockton CA 95213
Tel: 209.944.0836 ext: 151 Fax: 209.941.1744
***@***.*** www.wcmagnetics.com

ingo916, 2024-05-05:
[image]

ArcAIN6, 2024-05-07:
re-reading your initial post... both methods?? 

In you post, the procedure you describe is just installing, then uninstalling, then installing again. 

 ```
wget https://github.com/Z-Bolt/OctoScreen/releases/download/v2.7.4/octoscreen_2.7.4_armhf.deb  <-- Downloads the package
sudo dpkg -i octoscreen_2.7.4_armhf.deb <-- installs the package

and
wget https://github.com/Z-Bolt/OctoScreen/releases/download/v2.7.4/octoscreen_2.7.4_armhf.deb <-- Downloads the package
sudo dpkg -r octoscreen <-- Uninstalls the package
sudo dpkg -i octoscreen_2.7.4_armhf.deb <--- Installs the package
sudo reboot now <--- Reboots the system
```

You've also used the deploy script, since that is meant for creating an environment with multiple instances of octoprint on the same computer, i'm not sure octoscreen would work with that, as it's an interface for a single instance of octoprint. 
Quote from their repo's readme.md:
```These files provide a bash script for quickly deploying multiple octoprint instances on a single computer.```

If you only have 1 printer connected to a raspberry pi,  you're best bet is to flash an SD card with the OctoPi image, and then install octoscreen using the normal install method: 

 ```
wget https://github.com/Z-Bolt/OctoScreen/releases/download/v2.7.4/octoscreen_2.7.4_armhf.deb
sudo dpkg -i octoscreen_2.7.4_armhf.deb
```

If however you're connecting multiple printers, and running multiple instances of octoprint at once, i don't think octoscreen will work for you.

ingo916, 2024-05-07:
Thanks for the reply. This is a single instance for one printer. I will look at a different wat to install octoprint on a pi 5. I thought using the deploy script was the preferred method for installing on a pi 5.

ArcAIN6, 2024-05-07:
Oh... i'm kind of dumb... not sure why i didn't realize you were talking about a raspberry pi 5
OctoPi does not support the Raspberry Pi 5 yet

[image]

ingo916, 2024-05-07:
No worries

ingo916, 2024-05-09:
I installed OctoPrint on my Raspberry Pi 5 using this method https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspberry-pi-os-debian/2337

It is working perfectly; all the plugins I use work now except for OctoScreen. Is there a log file I can look at or provide to see what is causing OctoScreen to not run or be configured?

JeffB42, 2024-05-28:
> It is working perfectly; all the plugins I use work now except for OctoScreen. Is there a log file I can look at or provide to see what is causing OctoScreen to not run or be configured?

yes, in your config file, set OCTOSCREEN_LOG_FILE_PATH to the path of a text file, and set OCTOSCREEN_LOG_LEVEL to debug.

closed 2024-05-30 by ingo916