Z-Bolt/OctoScreen #379: Feature Request: Cura Preview on screen

ArcAIN6, 2024-05-05:
I'm currently using a 7" HDMI touch screen on my printer, and to be honest, that's a LOT of real estate for just a few buttons. 
I don't know if this has been done already ,and i'm just too dumb to find out how, but i think it would be nice if i could resize the buttons a little, and add the preview thats generated by cura when i send it to octoprint.  (I already have octoprint set up to show the previews in the file list / while printing, but having it on the actual LCD would be nice as well)

Thanks in advance, and i love this project!

JeffB42, 2024-05-28:
"... that's a LOT of real estate for just a few buttons..."

"...i think it would be nice if i could resize the buttons a little..."

Sorry, but the UI is fixed.  It expands in the sense that it takes up the entire screen, but not in the sense that one can add more buttons to the screen.  In general, OctoScreen has a 4x3 grid layout, and regardless of wether it is viewed on a 3.5" screen or a 7" screen, the layout is 4x3.

As far as a preview for Cura goes, do you have the "Cura Slicer Thumbnail" plug-in (in OctoPrint) installed?  I know previews show up when the "Prusa Slicer Thumbnail" plug-in is install, and previews should also work when the "Cura Slicer Thumbnail" plug-in is installed.

ArcAIN6, 2024-05-28:
Yes, the previews show up in octoprint, i just wanted the preview on the screen as well

JeffB42, 2024-07-12:
If you have the Cura Slicer Thumbnail plug-in is installed, they should show up in the "select file to print" screen.  (this was released in 2.7 and the notes for this are at https://github.com/Z-Bolt/OctoScreen/wiki/Version-2.7.  If they are not showing up there, then there's a bug.

On the other hand, if you are expecting a thumbnail to appear on the "currently printing"/print status screen, that is not implement and there isn't plans to do so.

closed 2024-07-12 by JeffB42