Z-Bolt/OctoScreen #380: Feature Request

J12345T, 2024-05-25:
Dear Developers. 

I have been using Cura for a few years mainly for 3d printed RC aircraft. This focuses on light weight foaming pla filaments printed in a single or double wall outer shell. The product is a light weight but often fragile model.

I have been experimenting with a 2 wall outer shell print where the 2 walls are artificially separated by a small amount. For the same weight of filament this gives a significant increase in strength to the model. 

I tried to trick the slicer by using the Outer Wall Inset feature and giving it a small negative value. Whilst the slicer accepted a value (-0.2mm) it did not seem to have an effect on the print. 

So I designed a CAD model with a 1mm shell. I set a 0.4mm wall width and the slicer printed an inner and outer wall separated by ~0.2mm. This is how I know the method produces a stronger result.

My request is to have a setting where I can specify a 'gap' between the outer wall and the inner wall such that I can CAD a solid model and let the slicer position the 2 walls as I've described.

A second feature request would be the ability to add a fuzzy wall finish to the INNER wall only. This once again would be to enhance strength for the minimum additional filament weight.

Hopefully I have explained my request adequately however, if not I am happy to provide more information. Thank you again for your fantastic work.

[image]

JT.

JeffB42, 2024-05-28:
Maybe I'm missing something her, but this sounds like a FR for Cura, not OctoScreen.

J12345T, 2024-05-28:
Sorry yes. A Feature Request for Cura.

This is the first time I've posted to Github so please forgive me if I've
got it horribly wrong. Any chance you could point me in the right direction?

Cheers,

James.

JeffB42, 2024-05-28:
I don't know, but maybe this - https://github.com/Ultimaker/Cura

closed 2024-05-28 by JeffB42

J12345T, 2024-05-28:
Thanks. Not sure how I got so far off track.